FunRobots/candybot_v2 #23: Не запускается rhvoice в контейнере docker

MeisterUrian, 2016-10-29:
1. установка rhvoice (детали - https://launchpad.net/~linvinus/+archive/ubuntu/rhvoice):
sudo add-apt-repository ppa:linvinus/rhvoice
sudo apt-get update

установка как клиент-сервер:
sudo apt-get install rhvoice rhvoice-russian rhvoice-english

установка как модуль speech-dispatcher:
sudo apt-get install speech-dispatcher-rhvoice rhvoice-russian rhvoice-english

=== Использование ===
1) Если вы установили пакет rhvoice то нужно выполнить команду
echo "Проверка синтезатора речи" | RHVoice-client -s Anna+CLB | aplay

2) Если вы установили пакет speech-dispatcher-rhvoice то нужно выполнить команду
echo "Проверка синтезатора речи" | spd-say -o rhvoice -l ru -e -t female1

Проблема: не работает **в docker контейнере** 
[image]

В реальной Ubuntu приложение spd-say использует сокет по адресу /run/user/1000/speech-dispatcher/speechd.sock , который (вместе с каталогом) создается в момент запуска spd-say.
Если передать этот сокет в контейнер:
sudo docker run -ti -v run/user/1000/speech-dispatcher/speechd.sock:/root/.cache/speech-dispatcher/speechd.sock full_rpo ,
то spd-say работает.
Предположительно проблема в работе сети внутри контейнера

closed 2017-03-25 by mnrozhkov

ilyar, 2017-10-20:
@MeisterUrian получилось запустить?

MeisterUrian, 2017-10-20:
@ilyar К сожалению, так и не получилось. Если передавать unix-сокет speech-dispatcher-а внутрь контейнера, тогда всё работает.
Получается, что внутри контейнера speech-dispatcher не может создать сокет.