FunRobots/candybot_v2 #76: fix in motion/body hardware bug

MeisterUrian, 2017-05-01:
[image]

MeisterUrian, 2017-05-28:
запуск motion-контроллера
rosrun coffebot motion_body_controller.py

запуск соответствующего теста
rosrun test_motion_body_controller.py

mnrozhkov, 2017-05-28:
Проблема в конструкторе Servo() !!! 

При подключении 1 PCA9685 не нужно указывать адрес в self.pca9685 = Adafruit_PCA9685.PCA9685(address)  
Более того, при создании объекта PCA9685() умолчанию используется адрес 0x40 

В случае подключения нескольких модулей PCA9685 пример можно найти здесь: https://learn.adafruit.com/16-channel-pwm-servo-driver/chaining-drivers   

TODO:
1. Исправить конструктор Servo() 
- убрать 'address' в создании объекта pca9685
- переименовать 'address' в 'channel'
2. В конструкторах классов Body  переименовать параметр _d_SERVO_ADDRESS в _d_SERVO_CHANNEL
3. То же сделать для  Eyebrows, Head
4. Внести изменения в тесты