FunRobots/candybot_v2 #98: Написать документацию по обновлению репозитория в docker

mnrozhkov, 2017-05-28:
Как обновить coffebot в докер? 

Я создал ветку fix-servo , внес изменения, хотел залить в RPi в docker, он ругается: 
[image]

Как лучше сделать?

MeisterUrian, 2017-05-28:
Т.к. каждая ветка - это отдельный пакет, то ее нужно клонировать  на уровень выше - в ~/caktin_ws/src
Но т.к. имя пакета coffebot должно сохраняться независимо от названия ветки на гитхабе, то для разных веток лучше держать разные рабочие папки catkin

Т.е. проще всего сделать следующее:
cd ~
mkdir -p catkin_ws_fix_servo/src (или любое другое имя папки)
cd catkin_ws_fix_servo
catkin_make
cd src
git clone https://github.com/FunRobots/candybot_vr.git -b fix-servo coffebot

Если попытаться в одном рабочем пространстве (одной папке) catkin создать несколько пакетов с одинаковыми именами в src/<имя_каталога_пакета>/src/<имя_пакета> (имена каталога и пакета могут совпадать). то ROS выдаст ошибку вследствие конфликта

mnrozhkov, 2017-05-28:
Мне кажется все проще, дело совсем не в ROS, а в настройке git config, чтобы можно было делать gut pull 

ROS даже не будет видеть, что что ты переключился в другую ветку git

MeisterUrian, 2017-05-28:
по сути ты хочешь слить 2 ветки, т.к. в одну пытаешься клонировать в другую
т.е. если считаешь, что локально эти ветки можно слить, то можно это и сделать
если сливать не нужно/нельзя, то надо создавать новое пространство catkin - в src может лежать только один пакет с одним и тем же именем

closed 2017-06-16 by MeisterUrian